TheAirBlow/HyperSploit #1: 10000

conflagration27, 2024-04-25:
[image]

I think my device can no longer be helped, right? I tried several settings.apk, it doesn't work with any =(

TheAirBlow, 2024-04-25:
> ![image](https://private-user-images.githubusercontent.com/167520015/325685134-7cf81f92-0062-43f7-b29d-629333f652bd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQwNzI1NTAsIm5iZiI6MTcxNDA3MjI1MCwicGF0aCI6Ii8xNjc1MjAwMTUvMzI1Njg1MTM0LTdjZjgxZjkyLTAwNjItNDNmNy1iMjlkLTYyOTMzM2Y2NTJiZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQyNVQxOTEwNTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yNjk3NTQzM2JhNmE5OWQxZmQxMzQ4MzQ1ZjMwNTYxNDU4MTc2MjY4NmM1MmRjZmZmNWQ2ZjY2Zjc2ZjM2N2M4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.QAt7Bhj2xQLHdtilNkqODTeUp-83sJ7npxOFXq6BVhg)
> 
> I think my device can no longer be helped, right? I tried several settings.apk, it doesn't work with any =(
> 
> but ty for hope

From what I've heard, yeah. You're pretty much screwed.

TheAirBlow, 2024-04-25:
Mainland China is very restricted atm, https://github.com/MlgmXyysd/Xiaomi-BootLoader-Questionnaire could be helpful if you'd like to unlock legitimately.

TheAirBlow, 2024-04-25:
Also, error 10000 is common for that one other tool because of how insanely lenient PHP is and was caused by failed decryption (due to a patched Settings app), but here it clearly succeeded and still failed, which is a bit weird...

Are you sure your device is from Mainland China?

conflagration27, 2024-04-26:
Yes, from mainland China. Just in case, I tried to choose "not from mainland China", the result was the same. Okay, it looks like I'll have to turn to paid unlocking services, thank you very much for the work done!

Orlandus, 2024-05-05:
Hello. Have you any news about xiaomi 14 ultra?
can you advise which one settings.apk to try?

closed 2024-05-11 by TheAirBlow