TheAirBlow/HyperSploit #3: Parameter app freezing and closing after downgrade

WWWhiteWolf, 2024-05-11:
Hi. I try your new tool, i think it's my last chance to root my device (Xiaomi 13T).
I previously had error #1000 from https://github.com/MlgmXyysd tool due to new cryptographic methods.

I tried to downgrade my android security app version using your tool but this cause the parameter app to bug and close itself.

I don't know if you've some informations on this topic ? Waiting your feddbacl, I succeed to restore the previous one.

TheAirBlow, 2024-05-11:
It depends on what HyperOS version you're on, and even on the device. There is no universal solution, at least for now.

I **might** add a feature that would automatically extract the settings app from an official firmware package, but that would be really difficult to implement from scratch - and have it work on windows (it doesn't support ex4)

WWWhiteWolf, 2024-05-11:
Hi again, sorry for double post. I overpass this issue by downloading another com.android.settings from https://www.apkmirror.com/apk/xiaomi-inc/settings-3/settings-3-14-release/settings-14-3-android-apk-download/
Hope it's a safe one. This release is older than the one in your release (september 2023) and is in english in my case (not in french).
Thanks this one I succeed bypassing the Xiaomi rooting validation thanks your program.

I have now to wait 360 hours (it means 15 days) to unlock m'y phone bootloader with mi flash soft.

Hope it will works.

I don't know if it is possible know to come black to the genuine settings apk or if I have to wait these 15 days ?

Thanks a lot for your program. Hope my feedback could help :-)

Have a noce day.

Best regards,
Antoine

TheAirBlow, 2024-05-11:
> Hi again, sorry for double post. I overpass this issue by downloading another com.android.settings from [...]

In theory, yes. They didn't change the heartbeat request, and in my case Xiaomi didn't even reset the timer once when I was testing this tool (making a bunch of binding requests, which Mi Unlock tells you to avoid)

I waited 14 days and I successfully unlocked my phone just 2 days ago, zero regrets so far.

closed 2024-05-11 by TheAirBlow

WWWhiteWolf, 2024-05-11:
> In theory, yes. They didn't change the heartbeat request, and in my case Xiaomi didn't even reset the timer once when I was testing this tool (making a bunch of binding requests, which Mi Unlock [...]

Really I thank you ! I need three complete days to succeed, finally finding your repository :-)

I'll reset the apk and keep in touch with you once final unlock will be performed.