TheAirBlow/HyperSploit #6: Downgrade failed

tomkosm, 2024-08-12:
I am getting the error bellow when running the script. Probably the blocked downgrading, when I try to downgrade manually I get  ```App not installed as package appears to be invalid.``` Any ideas?
```
Installing unpatched Settings APK to the device...
Unhandled exception caught, can't continue!
System.Net.Sockets.SocketException (32): Broken pipe
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.ThrowException(SocketError, CancellationToken) + 0x48
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.System.Threading.Tasks.Sources.IValueTaskSource<System.Int32>.GetResult(Int16) + 0x62
   at AdvancedSharpAdbClient.AdbSocket.<SendAsync>d__17.MoveNext() + 0x24a
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x1c
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xbe
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4e
   at AdvancedSharpAdbClient.AdbClient.<InstallAsync>d__30.MoveNext() + 0x67d
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x1c
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xbe
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4e
   at HyperSploit.Program.<Downgrade>d__5.MoveNext() + 0x384
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x1c
   at HyperSploit.Program.<Downgrade>d__5.MoveNext() + 0x679
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x1c
   at HyperSploit.Program.<Downgrade>d__5.MoveNext() + 0x7bd
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x1c
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xbe
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4e
   at HyperSploit.Program.<Bypass>d__4.MoveNext() + 0x935
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x1c
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xbe
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4e
   at HyperSploit.Program.<Main>d__3.MoveNext() + 0xc0b
```

TheAirBlow, 2024-10-01:
You're probably out of luck. Your settings app might have a bumped version index, which makes it impossible to downgrade.

TheAirBlow, 2024-10-01:
Try uninstalling updates, it may help.