TheAirBlow/HyperSploit #8: reports success, but apparently does not actually work

ossilator, 2024-09-30:
```
> ./HyperSploit-Linux
  _   _                                 ____            _           _   _
 | | | |  _   _   _ __     ___   _ __  / ___|   _ __   | |   ___   (_) | |_
 | |_| | | | | | | '_ \   / _ \ | '__| \___ \  | '_ \  | |  / _ \  | | | __|
 |  _  | | |_| | | |_) | |  __/ | |     ___) | | |_) | | | | (_) | | | | |_
 |_| |_|  \__, | | .__/   \___| |_|    |____/  | .__/  |_|  \___/  |_|  \__|
          |___/  |_|                           |_|
Welcome to HyperSploit v1.0 by TheAirBlow!
You chose 2107113SG codename vili!
Is this device from Mainland China? [y/n] (n): n
Open Mi Unlock Status and attempt to bind account
Disabling mobile internet, taking over!
Successfully decrypted arguments and headers!
Sending account bind request impersonating MIUI 14...
Phone was successfully bound to Xiaomi account, you can now use Mi Unlock!
Would you like to run HyperSploit on another device? [y/n] (n): n
```

but the unlock status still shows the registration button as if nothing changed, and the unlock tool fails at the second stage as before.

the device is an 11t pro, upgraded to hyperos 1.0.2. the tool downgraded the settings up successfully, and the device was rebooted.

TheAirBlow, 2024-09-30:
> ```
> > ./HyperSploit-Linux
>   _   _                                 ____            _           _   _
>  | | | |  _   _   _ __     ___   _ __  / ___|   _ __   | |   ___   (_) | |_
>  | |_| | | | | | | '_ \   / _ \ | '__| \___ \  | '_ \  | |  / _ \  | | | __|
>  |  _  | | |_| | | |_) | |  __/ | |     ___) | | |_) | | | | (_) | | | | |_
>  |_| |_|  \__, | | .__/   \___| |_|    |____/  | .__/  |_|  \___/  |_|  \__|
>           |___/  |_|                           |_|
> Welcome to HyperSploit v1.0 by TheAirBlow!
> You chose 2107113SG codename vili!
> Is this device from Mainland China? [y/n] (n): n
> Open Mi Unlock Status and attempt to bind account
> Disabling mobile internet, taking over!
> Successfully decrypted arguments and headers!
> Sending account bind request impersonating MIUI 14...
> Phone was successfully bound to Xiaomi account, you can now use Mi Unlock!
> Would you like to run HyperSploit on another device? [y/n] (n): n
> ```
> 
> but the device says "couldn't verify", and the unlock status still shows the registration button as if nothing changed, and the unlock tool fails at the second stage as before.
> 
> the device is an 11t pro, upgraded to hyperos 1.0.2. the tool downgraded the settings app successfully, and the device was rebooted.

Of course it would decline - this tool intercept the request Mi Unlock sends and modifies the MIUI version. Trying to use Mi Unlock again would result in the same error as before, and you might get your waiting timer reset by doing that.

What you should actually try instead is rebooting into fastboot and trying the Mi Unlock app to see if it works.

closed 2024-09-30 by TheAirBlow

ossilator, 2024-09-30:
your response makes no sense, so i think you misunderstood something. the second paragraph is additional info, and of course this was done _before_ attempting to register.

TheAirBlow, 2024-09-30:
> your response makes no sense, so i think you misunderstood something. the second paragraph is additional info, and of course this was done _before_ attempting to register.

This tool REGISTERS your device, it's not supposed to make Mi Unlock registration work on the device.
It intercepts the request your device sends, which registers your device.
So, you can just go ahead and download Mi Unlock and see how much wait time you got. That's it.

ossilator, 2024-10-01:
i know all this, theoretically. but that's not the same as actually understanding the process.

i now retried unlocking, and apparently the countdown has actually started at my last attempt, so things did in fact work. :man_facepalming: 

re-reading the README, it's obvious that FAQ 2 is _attempting_ to answer my question. it just miserably fails at it, because it mentions a technical detail without explaining the implications. there should be a second sentence "the registration has in fact worked, if the tool says so in the terminal". additionally, point 3.ii under "bypass" should mention where/how such failure would be indicated (error message in the tool's terminal output?), and there should be a point 3.iii saying something like "the settings app will show no indication that the device is already registered". this is obviously a completely abysmal user experience (so feel free to explicitly blame xiaomi if this is how things would be if the process succeeded normally. if it isn't, acknowledge that the tool is imperfect in this regard, so users are not confused).

thanks

TheAirBlow, 2024-10-01:
\> i know all this, theoretically. but that's not the same as actually understanding the process.
The tool literally tells you `Phone was successfully bound to Xiaomi account, you can now use Mi Unlock!`

\> and there should be a point 3.iii saying something like "the settings app will show no indication that the device is already registered"
It's not the tool's fault, that's how Xiaomi's registration process works. It does not check if you had registered already or not, as there is simply no reason to do so, as it's extra labour for a single message, which would not benefit the average nerdy user who will want to unlock their bootloader.

ossilator, 2024-10-01:
>  The tool literally tells you `Phone was successfully bound to Xiaomi account, you can now use Mi Unlock!`

yes, and the device tells me literally the opposite. you _do_ understand that it makes sense to explicitly address this contradiction in a way that that makes it easy to follow and discern from other possible outcomes without having seen them already, right?

ossilator, 2024-10-01:
look, everything is totally obvious once you've understood it. a good user experience requires the application and the documentation to tell a coherent, unambiguous story. users have better things to do than trying to resolve the apparent contradictions themselves.