TheAirBlow/HyperSploit #10: Can"t downgrade my settings

hungrymachine1, 2024-10-06:
[image]
[image]
Did all the tasks you said but didnt work (((Enable Install via USB
Enable USB debugging (Security Settings)
Near the end of the page, tap Reset to default values 5 times
After more options appear, disable Turn on system optimization (if it still fails, you may need to reboot)))))

TheAirBlow, 2024-10-07:
You have a patched settings app. You can't really do anything about it, but you can try removing all updates of the app. If it still doesn't work, you're out of luck.

closed 2024-10-07 by TheAirBlow

hungrymachine1, 2024-10-07:
> You have a patched settings app. You can't really do anything about it, but you can try removing all updates of the app. If it still doesn't work, you're out of luck.

How do i do that? there arent any options to remove the updates and tried downgrading the settings app manually but it keeps saying [app not installed as package appears to be invalid]

TheAirBlow, 2024-10-07:
> How do i do that? there arent any options to remove the updates and tried downgrading the settings app manually but it keeps saying [app not installed as package appears to be invalid]

There's a "Delete updates" option, if it's missing, [try this APK](https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/releases/download/1.0-fix/Settings_Redmi_marble_V816.0.5.0.UMRINXM.zip)

NotesOfReality, 2024-10-17:
> There's a "Delete updates" option, if it's missing, [try this APK](https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/releases/download/1.0-fix/Settings_Redmi_marble_V816.0.5.0.UMRINXM.zip) instead. If it doesn't work, you **are out of luck** and you can't use this bypass.

First of all thanks for this project and this valuable reply of yours.

I'd like to report I managed to easily install this version of the "Settings" app on a Poco F6 Pro (vermeer), here are my specifics :
* running HyperOS 1.0.6.0.UNKEUXM;
* I did it by [SanMer's PackageInstaller 1.1.7](https://github.com/SanmerApps/PI/tree/v1.1.7), running in the Shizuku working mode ("service version 540, adb");
* with [Shizuku 13.5.4](https://github.com/RikkaApps/Shizuku/tree/v13.5.4);

@hungrymachine1 you might want to try this I think.

Yuki1536, 2024-11-12:
> > There's a "Delete updates" option, if it's missing, [try this APK](https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/releases/download/1.0-fix/Settings_Redmi_marble_V816.0.5.0.UMRINXM.zip) instead. If it doesn't work, you **are out of luck** and you can't use this bypass.
> 
> First of all thanks for this project and this valuable reply of yours.
> 
> I'd like to report I managed to easily install this version of the "Settings" app on a Poco F6 Pro (vermeer), here are my specifics :
> 
> * running HyperOS 1.0.6.0.UNKEUXM;
> * I did it by [SanMer's PackageInstaller 1.1.7](https://github.com/SanmerApps/PI/tree/v1.1.7), running in the Shizuku working mode ("service version 540, adb");
> * with [Shizuku 13.5.4](https://github.com/RikkaApps/Shizuku/tree/v13.5.4);
> 
> I've run the tool and it unfortunately failed with error `30001`, whelp. I hope this helps someone else anyway.
> 
> @hungrymachine1 you might want to try this I think.

What's your device model?

`30001` means that your device has been forced to verify the account qualification by Xiaomi. You cannot bypass the HyperOS unlocking restrictions.

hungrymachine1, 2024-11-12:
> What's your device model?
>
> `30001` means that your device has been forced to verify the account qualification by Xiaomi. You cannot bypass the HyperOS unlocking restrictions.

I had already unlocked my bootloader with the official way. Currently using custom oss. I'm never gonna use this kind of bypass any more. I almost hard bricked my mom's phone with another Dev's bypass script. Man that was really intense. Somehow I fixed that and saved my life 😁. Not gonna spread hatred towards hypersploit but it's better to go through the official process. Just have to be quick to apply at time.

NotesOfReality, 2024-11-12:
> What's your device model?
> 
> `30001` means that your device has been forced to verify the account qualification by Xiaomi. You cannot bypass the HyperOS unlocking restrictions.

My device is a Poco F6 Pro (vermeer).

Bossburnz, 2024-11-15:
> > > What's your device model?
> > > `30001` means that your device has been forced to verify the account qualification by Xiaomi. You cannot bypass the HyperOS unlocking restrictions.
> > 
> > 
> > @Yuki1536 My device is a Poco F6 Pro (vermeer).
> 
> You device was first launched with HyperOS so you're out of luck. The 1st version of global device is `OS1.0.2.0.UNKMIXM`.
> 
> Mi Community unlocking application is required.

How about the Redmi Note 13 Pro 5G. I accidentally upgraded it to HyperOS and was unaware that unlocking the bootloader would be significantly more challenging.

Yuki1536, 2024-11-15:
> How about the Redmi Note 13 Pro 5G. I accidentally upgraded it to HyperOS and was unaware that unlocking the bootloader would be significantly more challenging.

Try out the bypass solution. If out of luck contact a paid flashing service to downgrade your ROM via EDL. (You download the fastboot ROM you preferred and they will auth and flash for you)

martinezix00, 2025-01-28:
> > > You have a patched settings app. You can't really do anything about it, but you can try removing all updates of the app. If it still doesn't work, you're out of luck.
> > 
> > 
> > How do i do that? there arent any options to remove the updates and tried downgrading the settings app manually but it keeps saying [app not installed as package appears to be invalid]
> 
> There's a "Delete updates" option, if it's missing, [try this APK](https://github.com/MlgmXyysd/Xiaomi-HyperOS-BootLoader-Bypass/releases/download/1.0-fix/Settings_Redmi_marble_V816.0.5.0.UMRINXM.zip) instead. If it doesn't work, you **are out of luck** and you can't use this bypass.

Xiaomi Pad 6S CN with HyperOS 2.0.3.0 - patch and changing Setting apps doesn't work either 

[image]
I just hope that one day there will be another exploit.....which will handle this. While waiting I reject all HyperOS updates...

Stromduster, 2025-03-08:
Is there a way to make it work on HyperOs 2 to expect ?

martinezix00, 2025-03-09:
> Is there a way to make it work on HyperOs 2 to expect ?

Currently there's no way to unlock hyperOS in other way than official channel. Xiaomi made the process impossible especially for newer devices :(

Kirolos124, 2025-05-15:
"After I saw version 10, I realized the problem I have has no solution.
The phone is Xiaomi 11 Lite 5G NE, Android 14, HyperOS 2.0.1.0 Global (Lisa).
I want to unlock the bootloader, but I can't because of some restrictions. Can you help me if there's a solution to this problem?

TheAirBlow, 2025-05-26:
Y'all can try the nuclear option in the [1.1.0](https://github.com/TheAirBlow/HyperSploit/releases/tag/1.1.0) update. It basically automates the update deletion step.

Mzafar-maalim, 2025-10-03:
[image]
 Os is patched,any idea?it's 2025