TheAirBlow/HyperSploit #17: Dont work on mi pad 5 (Ch)

Tim36tt, 2024-11-23:
So I decided to try to unlock via HyperSploit to unlock the Mi Pad 5 from the Chinese market
and in the end it didn’t unlock but showed this error

**Successfully decrypted arguments and headers!
Sending account bind request impersonating MIUI 14...
Error 20045: The place where the account is registered does not match the place where the phone is sold.**
What to do?

TheAirBlow, 2024-11-23:
It's the best kind of error - it's one that tells you what is the problem. And no, we cannot bypass it.

Try registering a new account with a VPN for the target country.

closed 2024-11-23 by TheAirBlow

Tim36tt, 2024-11-24:
ok i try

jorbig, 2025-01-06:
> Error 20045: The place where the account is registered does not match the place where the phone is sold.

Did this work for you? I'm running into the same issue...

jorbig, 2025-07-30:
@furkan5665st nope, I ended up using my device without unlocking the bootloader and rooting it... Shizuku fills most of my needs.