TheAirBlow/HyperSploit #33: Not working in ubuntu focal (20.04)

hucky58, 2025-02-16:
## Why another tool?
1) This tool is much more user-friendly, is a single file and has       **_zero dependencies_** 

No chance to run because

./HyperSploit-Linux: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by ./HyperSploit-Linux)
./HyperSploit-Linux: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by ./HyperSploit-Linux)

So there ARE dependencies ...

Ok, sources available but no howto compile for dummies.

TheAirBlow, 2025-02-17:
Of course it depends on GLIBC, for christ's sake... Every Linux program does. No way someone writes a program my going tons of syscalls manually.

Most probably your distro is too old and is doing you a disservice.

TheAirBlow, 2025-02-17:
To compile, you just need a .NET SDK of version 8 or above, and just run `dotnet publish`. That's it.

TheAirBlow, 2025-02-18:
I have only now noticed the title. `20.04` is ridiculously out of date. It was released 5 years ago, and [will soon be EOL](https://ubuntu.com/blog/ubuntu-20-04-lts-end-of-life-standard-support-is-coming-to-an-end-heres-how-to-prepare). I'd urge you to upgrade to `24.04`, which is the next LTS release.

closed 2025-02-18 by TheAirBlow