TheAirBlow/HyperSploit #35: Install Failed Version Downgrade

kuttahaitu, 2025-02-21:
i disable Turn on system optimization and reboot the phone multiple times but still these issues showing 
"AdvancedSharpAdbClient.Exceptions.AdbException: Failure [INSTALL_FAILED_VERSION_DOWNGRADE: Downgrade detected: Update version code 34 is older than current 35]

   at AdvancedSharpAdbClient.AdbClient.<InstallAsync>d__30.MoveNext() + 0x9f6
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at HyperSploit.Program.<Downgrade>d__5.MoveNext() + 0x2f4
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at HyperSploit.Program.<Downgrade>d__5.MoveNext() + 0x57b
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at HyperSploit.Program.<Downgrade>d__5.MoveNext() + 0x6ac
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at HyperSploit.Program.<Bypass>d__4.MoveNext() + 0x605
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at HyperSploit.Program.<Main>d__3.MoveNext() + 0xbc0"

I'm using Poco M6 Pro 5G running on Hyper Os version 2.0.3.0

LorierMatthieu, 2025-02-23:
Same for me

flourishers-edge, 2025-02-23:
same for me

TheAirBlow, 2025-02-23:
See #10

TheAirBlow, 2025-02-23:
And yeah you are 100% out of luck, you have patched firmware

closed 2025-02-23 by TheAirBlow

flourishers-edge, 2025-02-23:
> And yeah you are 100% out of luck, you have patched firmware

I manually tried to downgrade the settings app and then it worked but now got the timeout for 168 hours while unlocking is there any way to bypass that I know it's not the right place for this concern but just wanted the information thanks for the bypass

kuttahaitu, 2025-02-25:
> I manually tried to downgrade the settings app and then it worked but now got the timeout for 168 hours while unlocking is there any way to bypass that I know it's not the right place for this [...]

how, can you share that setting app

nicorac, 2025-10-19:
> I manually tried to downgrade the settings app and then it worked but now got the timeout for 168 hours while unlocking is there any way to bypass that I know it's not the right place for this [...]

168 hours delay was the expected wait time for MIUI users.
How were you able to downgrade the settings app?