TheAirBlow/HyperSploit #42: Error 86015: Invalid device signature

JSmaler, 2025-03-20:
Hey, was just trying this tool, and have run into the "Error 86015: Invalid device signature".

I _assume_ this _may_ be because I am on an older OS version (MIUI 14.0.11), but as the project does nowhere state what version is _required_, and older versions tend to be more permissive, I still tried.

If my assumption is correct, it would probably be good to note this version requirement directly in the readme.md. 

Below is the entire terminal output of the program running under fedora 41.
ADB and fastboot work on this system, and the program does appear to interface with the device correctly. I have not yet tried to run the app with sudo, as I do have some reservations about giving apps that don't need it sudo.
```
jkr@framework-16:~/Documents/dev/hyperSploit$ ./HyperSploit-Linux 
  _   _                                 ____            _           _   _   
 | | | |  _   _   _ __     ___   _ __  / ___|   _ __   | |   ___   (_) | |_ 
 | |_| | | | | | | '_ \   / _ \ | '__| \___ \  | '_ \  | |  / _ \  | | | __|
 |  _  | | |_| | | |_) | |  __/ | |     ___) | | |_) | | | | (_) | | | | |_ 
 |_| |_|  \__, | | .__/   \___| |_|    |____/  | .__/  |_|  \___/  |_|  \__|
          |___/  |_|                           |_|                          
Welcome to HyperSploit v1.0 by TheAirBlow!
You chose 23043RP34G codename pipa!
Is this device from Mainland China? [y/n] (n): n
Open Mi Unlock Status and attempt to bind account
Disabling mobile internet, taking over!
Successfully decrypted arguments and headers!
Sending account bind request impersonating MIUI 14...
Error 86015: Invalid device signature
Would you like to run HyperSploit on another device? [y/n] (n): n
```

TheAirBlow, 2025-03-21:
Yeah, that may be the case as another user reported that updating has helped.

Does it work out of the box? Can you dump your settings APK and send it here?

JSmaler, 2025-03-21:
I don't quite know what you mean with "out of the box". If you mean the function of the program, then yes.

As for the APK, I will try to dump it, but I have little know-how with android, in fact, I do all this to replace it, so this may take some time for me to figure out.

JSmaler, 2025-03-21:
As for the APK, turns out that isn't as hard as I feared, [here's the "MISettingsPad.apk"](https://workupload.com/file/y9gDW8wasbq).

TheAirBlow, 2025-03-21:
Sorry, by "out of the box" I meant if it works without the tool.

JSmaler, 2025-03-21:
Well, no.
If Xiaomi's systems worked, I would not have started to look for some way to circumnavigate them.

TheAirBlow, 2025-04-01:
The request has a ton of xioami account stuff and other shenanigans, like device signatures generated by the Security Center (from what I remember when I was reverse-engineering the whole thing locally)

Does it tell you to go to Mi Community or does it result in some other unexpected error?