TheAirBlow/HyperSploit #57: INSTALL_FAILED_INVALID_APK

Kirolos124, 2025-05-14:
Try downgrading to an unpatched version? [y/n] (n): y
Decompressing Settings.apk bundled within HyperSploit...
Installing unpatched Settings APK to the device...
Unhandled exception caught, can't continue!
AdvancedSharpAdbClient.Exceptions.AdbException: Failure [INSTALL_FAILED_INVALID_APK: Package com.android.settings are not updateable.]

   at AdvancedSharpAdbClient.AdbClient.<InstallAsync>d__30.MoveNext() + 0x9f6
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at HyperSploit.Program.<Downgrade>d__5.MoveNext() + 0x2f4
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at HyperSploit.Program.<Downgrade>d__5.MoveNext() + 0x57b
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at HyperSploit.Program.<Downgrade>d__5.MoveNext() + 0x6ac
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at HyperSploit.Program.<Bypass>d__4.MoveNext() + 0x605
--- End of stack trace from previous location ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw() + 0x20
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task) + 0xb2
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task, ConfigureAwaitOptions) + 0x4b
   at HyperSploit.Program.<Main>d__3.MoveNext() + 0xbc0
Would you like to run HyperSploit on another device? [y/n] (n):

TheAirBlow, 2025-05-15:
see #10

closed 2025-05-15 by TheAirBlow

TheAirBlow, 2025-05-15:
there's none.