TheAirBlow/HyperSploit #64: Wi-Fi Automatically Turns Off When Using Mi Unlock Tool During Xiaomi Account Binding

RainySorcerer, 2025-05-23:
When using the Mi Unlock Tool  during the Xiaomi account binding step required for bootloader unlocking, the tool automatically disables Wi-Fi by default. While this behavior may be acceptable on smartphones like marble (which have a SIM card and can fall back on mobile data), it causes a serious issue for tablets such as nabu, pipa, or Pad 7, which do not come with SIM support and rely solely on Wi-Fi for internet connectivity.

As a result, these tablets are unable to maintain a stable internet connection during the binding step, making it impossible to complete the bootloader unlock process.

Please look into this issue and consider updating the behavior of the tool to retain Wi-Fi connectivity for Wi-Fi-only devices such as tablets. This small change would ensure users can unlock their devices without unnecessary complications.

TheAirBlow, 2025-05-23:
will do if I find a way to detect if a device supports cellular data or not

RainySorcerer, 2025-05-24:
> will do if I find a way to detect if a device supports cellular data or not

Thx

TheAirBlow, 2025-05-26:
What output do you get when running these commands:
1. `adb shell pm list features`
2. `adb shell dumpsys isub`

TheAirBlow, 2025-05-26:
@RainySorcerer I want to make sure these actually work before implementing it. Both are good candidates.

TheAirBlow, 2025-05-26:
Fixed with commit e17d3f9

closed 2025-05-26 by TheAirBlow