TheAirBlow/HyperSploit #66: Unable to unlock my Redmi note 12 4G NFC (topaz)

ManBru11, 2025-05-26:
Hello:

I tried with the nuclear option, but it seems that I have a patchd version of hyperos (v.2.0.2.0VMGEUXM). Can you try to make a version that works with this version of hyperos?

Thank you very much.

[image]

TheAirBlow, 2025-05-26:
The patched Settings app uses a random 32 byte AES key, that is logged in an encrypted form with a 256 bit RSA public key, which is strong enough to be impossible to crack.

mrqx0195, 2025-07-26:
> The patched Settings app uses a random 32 byte AES key, that is logged in an encrypted form with a 256 bit RSA public key. It's strong enough to be impossible to crack. Only Xiaomi has the private key required for decryption.

Unfortunately, I'm also facing the same issue - my phone came with the patched system out of the box... I have some questions though: From which HyperOS version did this patch start? And if I downgrade the OS at an official Xiaomi service center, will HyperSploit work properly?

TheAirBlow, 2025-07-26:
> And if I downgrade the OS at an official Xiaomi service center, will HyperSploit work properly?

[image]

ManBru11, 2025-07-26:
Apparently my device is under warranty and the battery has swollen, so they will give me a new device, which will come with MIUI 14, which is easier to unlock. Although if it comes sealed in its box I will probably sell it.