TheAirBlow/HyperSploit #67: Failed, Please help me!

zhouhm, 2025-05-28:
```
# ./HyperSploit-Linux 
  _   _                                 ____            _           _   _   
 | | | |  _   _   _ __     ___   _ __  / ___|   _ __   | |   ___   (_) | |_ 
 | |_| | | | | | | '_ \   / _ \ | '__| \___ \  | '_ \  | |  / _ \  | | | __|
 |  _  | | |_| | | |_) | |  __/ | |     ___) | | |_) | | | | (_) | | | | |_ 
 |_| |_|  \__, | | .__/   \___| |_|    |____/  | .__/  |_|  \___/  |_|  \__|
          |___/  |_|                           |_|                          
Welcome to HyperSploit v1.1 by TheAirBlow!
You chose 22127RK46C codename socrates!
Is this device from Mainland China? [y/n] (n): y
Make sure you have a cellular connection with internet access!
Open Mi Unlock Status and attempt to bind account
Disabling mobile internet, taking over!
Failed to decrypt arguments and headers!
This probably means you have a patched Settings app.
Try downgrading to an unpatched version? [y/n] (n): y
Decompressing Settings.apk bundled within HyperSploit...
Installing unpatched Settings APK to the device...
Unhandled exception caught, can't continue!
System.Net.Sockets.SocketException (32): Broken pipe
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.ThrowException(SocketError, CancellationToken) + 0x3e
   at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.System.Threading.Tasks.Sources.IValueTaskSource<System.Int32>.GetResult(Int16) + 0x5c
   at AdvancedSharpAdbClient.AdbSocket.<SendAsync>d__17.MoveNext() + 0x279
--- End of stack trace from previous location ---
   at AdvancedSharpAdbClient.AdbClient.<InstallAsync>d__30.MoveNext() + 0xa69
--- End of stack trace from previous location ---
   at HyperSploit.Program.<Downgrade>d__7.MoveNext() + 0x740
--- End of stack trace from previous location ---
   at HyperSploit.Program.<Downgrade>d__7.MoveNext() + 0xd2f
--- End of stack trace from previous location ---
   at HyperSploit.Program.<Downgrade>d__7.MoveNext() + 0xe47
--- End of stack trace from previous location ---
   at HyperSploit.Program.<Bypass>d__4.MoveNext() + 0x37c
--- End of stack trace from previous location ---
   at HyperSploit.Program.<Main>d__3.MoveNext() + 0x8d8
Would you like to run HyperSploit on another device? [y/n] (n):

```

TheAirBlow, 2025-05-28:
your phone disconnected during the process. use a different cable

Jenna-66, 2025-06-14:
> your phone disconnected during the process. use a different cable

I had the same issue with different cables.
I fixed it by manual installing the Settings.apk from your repo.
I also had to log out from my mi-accout and login again.
But now I "only" have to wait a week and not (at least) a month.